SukkoPera/MegaDrivePlusPlus #3: Add Sega CD 2 support

SukkoPera, 2019-03-06:
This request from vincent sega could be implemented in MD++:

> I would like to modify my Mega Drive with an Arduino Nano and in addition i would like to modify my Mega CD 2 with Multibios.
> With Sebs Code it is possible to switch the different Bios via Pin8 and Pin10 of PIC16F630.
> Is it possible to perform the same with your code?
> If yes, which Pins of an Arduino Nano I have to use?

## Instructions
You are in need of a Eprom (27c2004) and burn all 3 bios Version into it. The usa, japanese and pal Version. You can select the different Bios Version via A16 (Pin 38) and A17 (Pin 39) of the Eprom.
The order is: USA-> JPN -> EUR.

You can do it by yourself. Take the CD-Bios Versions and make a byteswap each of them.

The concat the 3 Files via Windows Commandline copy

copy /B usa.bin + pal.bin + jpn.bin switchless.bin

The final goal is to get 2 wires from the 16F630 PIC to the Eprom.
Here we make use of 2 wires from the Sega Expansionport.

 

Pin8 yellow 16F630->Exp Port Pin A01@Genesis 2->Exp. Port Pin A30@SegaCD2->Pin 38Eprom

Pin 10 blue 16F630->Exp Port Pin A30@Genesis 2->Exp. Port Pin A30@SegaCD2->Pin 39Eprom

Pin A01 and A30 of the Expansionsport carrying GND. There are many ground Signals at the Expansionsport. So we can cut these two of and make use of the free Pins for the Bios select.

  | PIC 8 /Eprom 38 | PIC 10 /Eprom 39
-- | -- | --
EUR | 0 | 1
JPN | 1 | 0
USA | 0 | 0

## Considerations
But if I understand correctly, all you need is two pins that change theire value together with the region, correct? In that case you can easily modify the code, adding a couple of pinMode() and digitalWrite() calls that do what you need. By looking at the Nano pinout I see that D7 and D8 are currently unused, so you can use those pins (or A4/A5 if you moved those over).

hadess, 2019-05-08:
The way I understood it, the current best practice is to replace the builtin BIOS with one of [l_oliveira's region-free BIOSes](https://assemblergames.com/threads/region-free-bios-for-mega-cd-version-2-00.54254/).

rfancella, 2024-08-20:
> This request from vincent sega could be implemented in MD++:
>
>
> ## Instructions
>
> You are in need of a Eprom (27c2004) and burn all 3 bios Version into it. The usa, japanese and pal Version. You can [...]

Hello,
I would like to try integrating the three separate bios files into my Sega CD Model 2.  I understand the best practice is to use a region-free bios.  But I would like to see the three different boot animations depending on the region selected.  The OG instructions say to use a 27C2004.  I cannot find one anywhere and little documentation on it.  I understand that I need a 4m eprom.  I have used a 27C1024 for a single bios and it works just fine.  What is a good substitute for the 27C2004?  Maybe a 27C400, 27C4096 or 27C4002?  They seem to have comparable pinouts and I have some 27C400's.