SukkoPera/MegaDrivePlusPlus #5: Region is set too slowly at power-up

SukkoPera, 2019-05-04:
Some games do not start since they find the wrong region value (e.g.: Virtua Racing).

See: https://assemblergames.com/threads/megadrive-new-switchless-region-igr-mod.61273/page-10#post-934050

hadess, 2019-05-24:
Could this make a difference? https://github.com/Optiboot/optiboot/

That'd mean nothing to change in MegaDrive++ just a hardware/bootloader change on the target device.

SukkoPera, 2019-05-24:
Actually I already use Optiboot on all my Nanos, and it should be the default bootloader on current boards (i.e. those bought in the last year or so), so I'm not sure that makes any difference.

I think that some fuse values might fix the issue, but I need to test.

SukkoPera, 2020-01-13:
I have been taking some measurements with a logic Analyzer:

- Using the "old" Arduino Nano bootloader:
  [image]

  The time from power-up to the language/videomode lines reaching the appropriate values is ~1.46 seconds. Probably the bootloader has a long delay before actual code is started.
  It then takes ~0.45 s before the initial reset pulse is sent.

- Using the current Arduino Uno bootloader (Optiboot):
  [image]

  Power-up to lines settling is ~74 ms, from then to reset pulse it's ~100 ms more.

The Uno bootloader starts executing MD++ code WAAAAY sooner than the Nano bootloader (which is deprecated and should not be pre-flashed any longer on original Nanos, who knows about clones), so it is definitely recommended.

I still cannot say if 74 ms is fast enough to bypass the initial region check by most games. It surely works well on my Virtua Racing JAP, but even the slower Nano version worked fine. I think another game was mentioned in the original forum thread (which is now inaccessible), Ecco The Dolphin, but I cannot remember what region it was that was causing problems and I don't really wanna buy them all.

It's not clear to me why the time from the lines settling to the reset pulse also changes drastically with the two different bootloaders.

Also, the LA revealed some weird stuff going on at some power-ups. I had captured one but then hit the wrong button :(.

One last thing: some 6-button pads yield weird results at power-up, which might trigger mode-switching combos. Adding debouncing might help.

SukkoPera, 2020-01-13:
The 0.45s from lines settling to the reset pulse might be due to a spurious capture or something weird. This other capture took the usual 100 ms:
[image]

SukkoPera, 2020-01-22:
Just noticed [we are actually forcing the 100 ms delay](https://github.com/SukkoPera/MegaDrivePlusPlus/blob/experimental/MegaDrivePlusPlus.ino#L524) before the initial reset, which sounds totally useless now, so it could be removed.

mylogon341, 2020-06-30:
Hey. I've sort of come across this. I performed a mod for a client and they've reported issues with Sonic 3 not booting because of region restrictions. The weird thing is, the console is a PAL board (if I remember correctly) but the pal cart is coming up with with an error on boot saying that it will not work on NTSC consoles. Something like that. Either way - is this the same issue as this? If not, it's looking like I might have to make this a return job to repair the issue (if I can). 
Thanks

SukkoPera, 2020-07-01:
I really can't say, I think no one ever reported any issues with Sonic 3. What bootloader does that board use?

SukkoPera, 2021-11-03:
I have just got an email from the guy who was experiencing the issue on the AG forum thread (Thanks Chris!) saying:

> A few days ago I was going through my pile of unfinished projects and
came across the MPP. I decided to look it up and saw there was some more
development done, and that a new bootloader was available for the Nano.
>
> After flashing the bootloader and the latest master branch of MPP, it
works! Thank you so much for developing this.
>
>I can also confirm that it allows my JP Mega Drive Model 1 to work with
a U.S. Sega CD Model 2.

He also reports:
> I tried U.S. Sonic 3 and it seems to detect the region settings correctly for me (won't start in JP mode, starts fine in U.S.)

So I guess we can finally close this issue.

closed 2021-11-03 by SukkoPera