SukkoPera/MegaDrivePlusPlus #10: Something wrong??

Epitaph78, 2019-06-01:
Hi. I've installed this in my Mega Drive 2 with an Arduino Nano, verified all the connections were correct with a multimeter, but when I turn the console on, it seems to be locked in 60hz mode, the reset button does absolutely nothing at all, the START+DOWN combination changed the colour of the LED from green, to blue, to red, and back to green, and that's all it does. Even changing the mode and then turning the console off and on changes nothing. I find it weird the button combination is different to what it's supposed to be, and as I said, nothing at all changes on the screen. Would appreciate any insight.

SukkoPera, 2019-06-03:
Did you close the ticket because you have already solved the issue?

Epitaph78, 2019-06-03:
Yes. I was using a schematic from Sega-16 which showed the relation between
the colour coding and the arduino pinout, but it seems that pinout was an
old one as I later noticed the pinout in the sketch was different. So it
was a case of changing the wires around on the arduino, and it worked.

SukkoPera, 2019-06-03:
This made me realize there's no official wiring diagram on the wiki. Reopening so that I remember to fix that.

Epitaph78, 2019-06-03:
👍👍

SukkoPera, 2019-06-05:
[Installation page](https://github.com/SukkoPera/MegaDrivePlusPlus/wiki/Installation) revamped, hopefulyl everything will be clearer now.

closed 2019-06-05 by SukkoPera