SukkoPera/MegaDrivePlusPlus #13: 6 button controller not working properly on Model 2 Genesis

therealinfidelity, 2019-10-13:
Years back on my model 1, I installed the Arduino Nano and had no issues with my 6 button controller. I was able to use all buttons with no issue, even my 32x games functioned properly. My main attraction to this mod was the in-game-reset, which worked perfectly on my model 1/32x combo.

Now I've been out of the scene for this mod for some years, and I reinstalled my board into a model 2 genesis, and used the latest wiring diagram for the arduino nano, and obtained the most recent files, the mega++ & the readme file. Everything flashed correctly, but I'm running into issues in regards using an official wired 6 button controller.

The C button (using Sonic 1 as an example) works very poorly. Sonic doesn't even do a full jump. My start button doesn't seem to respond quickly. And with some 32x titles, I experience the same issue with my C button. But, using my original 3 button controller, everything is fine, buttons all operate the way they are supposed to.

So my question is, am I flashing the wrong files, and soldering incorrectly? Plesse help, cause this mod was PERFECT for me years ago, and I'm disappointed with the results I'm getting now with my model 2 genesis 6 button. infidel80@hotmail.com

SukkoPera, 2019-10-13:
Please check out [the FAQs](https://github.com/SukkoPera/MegaDrivePlusPlus/wiki/FAQ-and-Troubleshooting), number 1 in particular.

I suspect a bad solder joint, possibly a short on controller port pin 9, please double check.

therealinfidelity, 2019-10-13:
IGR always worked for me on every game I threw at it. I don't have any bridges on my control solder pins, or on my arduino nano. If that was the case then my 3 button controller wouldn't produce the proper C button function. So I'm assuming it has to be a firmware issue with the nano. Am I supposed to flash the "legacy" branch to the nano, and solder to the nano accordingly that the legacy version demonstrates?

SukkoPera, 2019-10-14:
As FAQ#1 says, MD++ does not interfere with the signals traveling between the controller and the console, it only "listens" to what's in there, thus it CANNOT be a firmware issue.

It CAN be a hardware issue, that's why I suggest double checking your solderwork.

On option could be REMOVING the L led on your Arduino, that is connected to pin 13 and thus might be causing the issue.

closed 2019-10-14 by SukkoPera

therealinfidelity, 2019-10-16:
L led removal did the trick. 6 button all set.