SukkoPera/MegaDrivePlusPlus #15: Random resets with 8BitDo controller

SukkoPera, 2020-02-01:
Ruslan Zakirov reports:

> I settled MegaDrivePlusPlus on my own MD2 VA1 PAL and found the
> following problems using the controller from 8BITdo. The fact is that
> when I press the combination A + start or B + C or A + B + C, this leads
> to a reset of the console in any game . But if I use a wired controller,
> then there are no such problems.

SukkoPera, 2020-02-01:
I'm pretty sure this is due to the 8BitDo receiver ouputting 3.3V instead of 5V as it should. I have already investigated this with the help of a friend due to the same receiver not working with OpenC64MegaDrivePadAdapter. He measured 3.25V when the button is not pressed.

I don't really want to cope with shortcomings due to bad designs by other people, but I can add some way of turning MD++ off temporarily.

SukkoPera, 2020-02-01:
I have added a disable function to the [experimental branch](https://github.com/SukkoPera/MegaDrivePlusPlus/tree/experimental): if the console is powered up while <kbd>Down</kbd> is held, combo detection will be disabled and IGR and mode switching from the controller will no longer work.

Please test and report.

chulpanovich, 2020-02-01:
> I'm pretty sure this is due to the 8BitDo receiver ouputting 3.3V instead of 5V as it should. I have already investigated this with the help of a friend due to the same receiver not working with [...]

Yes. I also noticed a difference of 3.3 and 5 volts.

chulpanovich, 2020-02-01:
> I have added a disable function to the [experimental branch](https://github.com/SukkoPera/MegaDrivePlusPlus/tree/experimental): if the console is powered up while Down is held, combo detection will be disabled and IGR and mode switching from the controller will no longer work.
> 
> Please test and report.

Yes thank you. I will check back soon and let you know.

chulpanovich, 2020-02-02:
> Yes thank you. I will check back soon and let you know.

I checked it and it works great. But each time you turn on the console, holding the button (down) is not very convenient. It would be great if this parameter was stored in memory, but then the control priority should be transferred to the reset button of the console itself. For example, press the reset two or three times and thus enable or disable the IRG.

chulpanovich, 2020-02-02:
I noticed that when you turn on the receiver 8bitdo it automatically turns off the IRG, but I plan to change the coordination of 3.3 and 5 volts using 74LCX245 and so I would like to manually control the switch.

SukkoPera, 2020-02-02:
That is too much work which I don't really want to do just because someone decided they were too lazy to respect the standard voltages :(.

Anyway on second thinking I don't think the 3.3V is the problem. Anything over 2.5V should be recognized as HIGH. Maybe it's the voltage when the signal is LOW? Can you measure it? But it doesn't make much sense either :(.

chulpanovich, 2020-02-02:
> That is too much work which I don't really want to do just because someone decided they were too lazy to respect the standard voltages :(.
>
> Anyway on second thinking I don't think the 3.3V is the [...]

Yes, this is really strange. The fact is that I have MD2 VA1.8 ASIA and it works great at 3.3 volts. I can not understand what is the difference with those consoles that have problems (

Ok. please tell me which parameter the IRG is turned off and I myself will add a command for the reset button. :-)

SukkoPera, 2020-02-02:
You can just set `enabled = false`, see https://github.com/SukkoPera/MegaDrivePlusPlus/blob/experimental/MegaDrivePlusPlus.ino#L590.

chulpanovich, 2020-02-02:
thank you very much!!!

chulpanovich, 2020-02-04:
the problem is not in the drops of 3.3 and 5 volts
I installed the chip 74HC245 and the levels became 5 volts, but the problem with 8BitDO remained (
The fact is that I have MD2 VA1.8 ASIA and it works great at 3.3 volts.

chulpanovich, 2020-02-04:
I also checked the receivers from KriKZZ Joyzz they also have 3.3 volts, but they work correctly. (

SukkoPera, 2020-02-04:
Can you measure the voltage when buttons are pressed? It's supposed to be ~0V.

chulpanovich, 2020-02-05:
I checked. I turned on the game SONIC 1 and when I pressed the buttons on all contacts 1,2,3,4,6,9 except 7, the voltage changed from 5 to 0 volts. Let me remind you, I measured the voltage after 74HC245

chulpanovich, 2020-02-05:
Maybe this is due to the speed of the survey, and somehow it is possible to change? I do not know, therefore I ask). I also noticed that when I press B+C then the command is executed as if I pressed left + start + C :-(

SukkoPera, 2020-02-05:
Naaah. I really don't know then. I don't have this receiver and I don't need to buy it, so unless someone can debug this on their own, I can't do much :(.

TurBoss, 2020-02-08:
Hello I successful wired the arduino nano to a IC BD M5 PAL VA4 

with the original 3 buttons gamepad it works like a charm Perfect

but with the 8bitdo there are issues...

restart ,region and Hz works

but the C button only works when holding START :/

i didn't manage to reproduce the OT issues about random restarts

My pad is a 8bitdo M30 2.4G 81CA

Thanks

TurBoss, 2020-02-08:
can't figure why but hosting start is not required anymore to be able to press C

chulpanovich, 2020-02-10:
I seem to have fixed the problem with the receiver 8BITDO. I replaced the resistors R68 and R68 with 10 ohms.

SukkoPera, 2020-02-10:
This is great! But you mean R68 and...? :)

Also is it on the receiver or on the console? Do all revisions need this in the latter case?

chulpanovich, 2020-02-10:
it's on the console R68 and R69.  I checked on a MD2 Europe 3pcs.  I had no such problem on the MD2 Asia model.  I will check in the future on different revisions of MD1 and MD2.

chulpanovich, 2020-02-10:
I used the source code to make my version of the mod for megadrive
now there are following functions:
- Switching regions Europe (EUR), America (USA), Japan (JAP)
- Switching frequency 50 / 60Hz
- Switching color standards PAL, NTSC, PAL60, NTSC4.43
- The image is always in color (two generators)
- Overclocking the central processor up to 10 MHz
- Switch sound filtering (when installing a supported MEGAAMP mod)
- Management of settings with the RESET button (no toggle switches)
- Manage settings from the gamepad with specific button combinations
- Light and sound indication of operation and set device parameters
- Non-volatile memory for saving parameters
- Disabling IGR and combinations if they are not needed
- Compact size 54 * 44 mm

SukkoPera, 2020-02-10:
Cool :). Remember that if you sell (or somehow share) your version you also need to offer the source code.

chulpanovich, 2020-02-10:
> Cool :). Remember that if you sell (or somehow share) your version you also need to offer the source code.

I remember:)

chulpanovich, 2020-02-13:
Hello. Unfortunately, replacing resistors does not work very well on some systems, and random combos reappeared. Maybe there is some way to make a timer for combos? For example, we press a combo and it only works two seconds after holding the buttons? Could you make such an option for customization?

SukkoPera, 2020-02-13:
I think you can exploit the debouncing function I've recently introduced in the *experimental* branch for this. Just increase [DEBOUNCE_BUTTONS_MS](https://github.com/SukkoPera/MegaDrivePlusPlus/blob/experimental/MegaDrivePlusPlus.ino#L175) to 2000U.

monsterro, 2020-02-13:
I'm speaking from memory and I can confirm when I'm back home. Had the same problem and for fun scoped pad inputs with 8Bitdo M30 connected. Rise and fall times were much worse than with official controller. The readpad routine was too fast so I delayed it with several 'nop' instruction, 8 if I remember correctly. It pretty much fixed the issue for almost every game I tried but I think Batman Forever still had issues, probably game was polling too fast. Anyway, I'll get my MegaDrive and play with it a bit tonight.

chulpanovich, 2020-02-13:
I checked it helped) I set the value to 250U. In Batman Forever also no problems were found. Maybe someone else will check and say here.

monsterro, 2020-02-13:
Button pressed on M30 (yellow) and official wired controller (blue):
[image]

Select (yellow) + button (blue) on M30
[image]

Select (yellow) + button (blue) on official wired controller
[image]

SukkoPera, 2020-02-14:
This is pretty weird. I used to think games read the lines right after (i.e.: a few tens of nanoseconds) setting Select low. That's why I had to write the ISR in assembly and try to be as fast as possible.

How can this be so slow and still be compatible with all games?

chulpanovich, 2020-02-15:
> I checked it helped) I set the value to 250U. In Batman Forever also no problems were found. Maybe someone else will check and say here.

Oh, now combos with these parameters do not work on megadrive 1 VA6 ((

monsterro, 2020-02-16:
MegaDrive is most likely using TTL voltage levels, which is minimum 2V for high, maximum 0,8V for low. ATmega328p running on 5V has minimum 3V for high, maximum 1,5V for low. 

Dotted lines below are set at 1,5V an 3V. Yellow impulses show when the button state is being read. Reading routine toggles one high and low before and after reading the controller. Both times shown the button is read as pressed. Delaying the read slightly allows the button presses to be read correctly.

[image]

@chulpanovich Can you try 74**LS**245 as a level shifter instead of 74HC245?

chulpanovich, 2020-02-16:
Yes, I will try 74ls245, but in my city there are no such chips, I will place an order and after a while I will tell you about the results.

monsterro, 2020-02-16:
I've edited my last reply. Try HCT instead LS.

chulpanovich, 2020-02-16:
But one circumstance becomes clear why, at different revisions of the consoles, the sensitivity to combos is different.

chulpanovich, 2020-02-17:
> I've done that by adding 10 nop instructions in readpad.S, just before **R_PORT, PAD_IOPORT**,

Please help, I did not quite understand what this means ((
Can I have an example? )
Today I look at the nearest store 74HCT245.

monsterro, 2020-02-17:
Like this:

[image]

chulpanovich, 2020-02-17:
6 nop end 150U give good results

Oj5000, 2020-04-14:
I've recently installed this into my megadrive 1 VA4 and had trouble with my clone arduino - the same as TurBoss where C only worked if you had Start pressed. I found the issue to be related to the RX0 input on the arudino sucking away some of the power from the controller port (I assume anyway) causing the effect.
Since I was only interested in having in game reset functionality I disconnected RX0, D7, D6 and D5 then connected controller port 9 (Start/C) to D5 and modified the buttons function to:

	word buttons = ((b1 & 0x80) >> 5) | (b1 & 0x08)
               | ((b2 & 0x80) >> 6) | ((b2 & 0x08) >> 3)
	       ;

I also removed lines 23-25 from readpad.S :
    andi R_STATE, 0xF0          ; Check if U/D/L/R are all low at the same time (This sets SREG flags)
    breq have6btn               ; If == 0, U/D/L/R are all low, this means we have a 6-button pad

I'm not sure if that mattered or if I had a short somewhere causing the reset to not function so don't trust this bit too much!

I also tried changing the reset combination to X,Y,Z and START but this only worked in 6 button compatible games, hence going with the 3 button approach.

I hope this is useful to anyone experiencing the strange START + C issues.

SukkoPera, 2020-10-04:
The workaround is now in the master branch. Please open a new issue if any new problems surface.

closed 2020-10-04 by SukkoPera