SukkoPera/MegaDrivePlusPlus #19: I have different model board.

MasterSAN23, 2020-09-10:
Hello. I have different model board then in list. Can you help me to install this mod to my board? Thanks.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SukkoPera, 2020-09-10:
Sorry but I have no idea on how to help. I'm adding the "help needed" label, maybe someone will chime in.

MasterSAN23, 2020-09-11:
Sad(( This is interesting project.