SukkoPera/MegaDrivePlusPlus #21: My PAL MegaDrive 2 is not in the list (PC BD MD2 VA0 PAL)

lorecast162, 2020-11-02:
Hello. My PAL Mega Drive 2 (PC BD MD2 VA0 PAL) is not in the list.

Sorry in advance if the pics aren't good enough but they're as good as I can take them right now.

[image]
[image]

lorecast162, 2020-11-02:
I suspect it to be similar to the VA1 but I'm not sure

SukkoPera, 2020-11-02:
Yes, as far as I can say it looks exactly the same. Try and let me know.

lorecast162, 2020-11-02:
I can't perform the mod yet, it's my only mega drive and unless I'm sure it's ok I'm not too inclined to doing it. Hope you understand.

lorecast162, 2020-11-02:
Tbf, it's not much about not being _able_ to do it (I'm sorta confident enough in my skills) but more about not really _wanting_ to take the plunge and do it

SukkoPera, 2020-11-02:
Sure, I understand, but I cannot help much, unless someone donates this board exactly.

Tagging as *help needed*.

chulpanovich, 2020-12-16:
> Здравствуйте. Мой PAL Mega Drive 2 (ПК BD MD2 VA0 PAL) отсутствует в списке.
> 
> Заранее извиняюсь, если фотографии недостаточно хороши, но они настолько хороши, насколько я могу их сделать прямо сейчас.
> 
> [image]
> [image]

This is how you can do it.
[image]
[image]

SukkoPera, 2020-12-17:
Thanks a lot @chulpanovich! Much appreciated!

@lorecast162 IF you happen to try this out, let me know if it works so that I can add it to the official docs.

pseiler, 2021-11-26:
I'll try this mod soon. But it'll take some time. I need to first flash the new/faster arduino bootloader etc. Also I take my time.
I will also add a picture of the finished mod.

pseiler, 2022-01-26:
The tutorial defenitely also working for the VA0 console. I successfully modded my Mega Drive VA0 PAL.
The cable management is trahs thats why I have no proper pictures.

Only this for reference
https://circuit-board.de/forum/index.php/Attachment/182409-PXL-20220118-120130224-jpg/

But the modchip works as expected.
Had to adjust the LED values to use a duoLED.
Maybe just mention this is in the Wiki.

Thanks for having this awesome option to mod a megadrive.
Mine works flawlessly with a DFO to generate a proper 60Hz signal.

SukkoPera, 2022-02-07:
Done, thanks a lot for your report!

closed 2022-02-07 by SukkoPera