SukkoPera/MegaDrivePlusPlus #22: MD2 VA3 USA board

RigoHoward, 2021-05-10:
Hello, first of all this modification its awesome so many props for this!!

Im interested to modify my MD2 VA3 NTSC console with it, but i see there isnt an specific diagram for it.

Here are some.pictures of the board for reference

[image]
[image]

I see some coincidences with the MD2 diagram on the wiki, but the cutting points it seems there are not the same. 

Thank you again in advance and im hoping that the mod would be possible on my console :)

chulpanovich, 2021-05-12:
[image]

SukkoPera, 2021-05-12:
That is great, thanks a lot!

@RigoHoward Please test and report!

RigoHoward, 2021-05-12:
All right guys that was fast, i will do the modification and write back about the results ;)

RigoHoward, 2021-05-13:
Okay, here is my installation (not tested yet)
[image]
[image]
[image]
[image]
[image]

RigoHoward, 2021-05-13:
@chulpanovich do i need to desolder the reset button and lift the legs?

chulpanovich, 2021-05-14:
> @chulpanovich do i need to desolder the reset button and lift the legs still?
In your case, you do not need to unsolder the button and raise your legs if you cut off the track.