SukkoPera/MegaDrivePlusPlus #24: Megadrive 1 - JAP VA4 not working

VincentNLOBJ, 2021-09-26:
Hi all,

I've acquired a modded MD JAP VA4 with ugly switches and decided to upgrade with MD++. I'm not interested on LED changing colour so I've not touched that part.

Following dragoncity scheme: https://dragoncity17.files.wordpress.com/2015/03/md-jap-va4.jpg
My board had JP3 / JP2 traces cut, so I have reconnected them with 2 wires below the board, then cut JP1 / JP4 traces.

After wiring up a this is what I've got:

[image]

Arduino powers up, and C button lights up arduino led, which is a good sign, but that's all about it.
Pressing reset button yield no effect (no reset or change language/hz when pressed).

For this reason I'm thinking to have followed a wrong scheme or there is a sort of incompatibility with the reset-in/out on the board.

Anyone have a working scheme of JAP VA4 board?

Cheers
-VincentNL

SukkoPera, 2021-09-26:
Why did you reconnect JP2/3? All JP1/2/3/4 must be cut!

VincentNLOBJ, 2021-09-26:
I assumed dragoncity scheme white straight lines meant they had to be connected! but that's an easy one, just removed the connection.

Regarding JP2 left contact seems in a pretty bad shape. 
Multimeter is dead when testing it against all the other JPs, not sure if it's normal, but it's worth to mention.

[image]

SukkoPera, 2021-09-27:
Did you cut the trace between Reset In and Out?

Try [enabling debug messages](https://github.com/SukkoPera/MegaDrivePlusPlus/blob/master/MegaDrivePlusPlus.ino#L194) and check what the serial monitor tells you.

VincentNLOBJ, 2021-09-28:
Yeah I did cut the reset in-out trace, following up dragoncity scheme.
When arduino gets 5v, reset button is disabled so unless it is solved that prevents the mod from working.

I'll try out debugging as you have suggested, thanks!

shanotype, 2023-05-11:
Hey guys I'm having the same exact issue the reset button doesnt do anything. Any ideas?

VincentNLOBJ, 2023-05-11:
No, honestly after having made 3 PAL machines working,  JAP VA4 is clearly
not compatible and requires debugging to find out why.

SukkoPera, 2023-05-11:
What does the debugging output say? Can you also check whether the reset button is active-low?

VincentNLOBJ, 2023-05-11:
I'm sorry but I have not access to that unit anymore.

shanotype, 2023-05-11:
I got it working. :)

After looking at PAL VA4 instructions for troubleshooting I noticed it was pretty similar enough and had the trace cut on the reset out end so decided I would join the reset in trace back and cut from reset out side but when removing the wire the whole trace lifted not sure if it got damaged when I soldered the wire to it I should have used thinner wire.

Anyway so I found where the trace lead to and soldered reset out to that, rejoined reset in, and already had a cut from the trace lifting and now everything works.

So to sum it up for people in the future on this model try cutting the same way as PAL VA4 picture. Shame VincentNLOBJ does not have it anymore so we could see if it fixes his same issue. Sorry for the lengthy post guys.

SukkoPera, 2023-05-11:
This is good news, thanks for reporting back!

VincentNLOBJ, 2023-05-11:
In case it does not bother you, I'd love to check out pics of the mod you
did. If I'll get the same unit again, will try it out.

Cheers!

shanotype, 2023-05-11:
It's messy atm will probably tidy it up some more. Also noticed your missing that Surface Mount between in and out dunno if that matters.

[image]
[image]
[image]
[image]

VincentNLOBJ, 2023-05-11:
Thanks really appreciate it!

SukkoPera, 2024-03-02:
Link added to Wiki, thanks for your help!

closed 2024-03-02 by SukkoPera