SukkoPera/MegaDrivePlusPlus #25: Pinout for IC M5 USA VA2 or PC BD MD2 VA2.3 USA

sincere360, 2022-01-05:
Hello I would to know if someone could help with the pinout on one of these two boards. The IC M5 USA VA2 or PC BD MD2 VA2.3 USA I have both systems and would like to mod one. Thank You

sincere360, 2022-01-05:
I was looking at the diagram for IC BD M5 PAL VA4 and all the points looks the same. If someone can confirm that would be great. I'm going to go and test with the this diagram.

sincere360, 2022-01-07:
I finished the mod and I am able to play the only PAL game I have. One thing I noticed is when I use the reset switch to change modes the color changes. I'm using an RGB LED so I get red, blue and green modes. I didn't see anywhere on the wiki what color indicates what region, maybe I missed it somewhere. Also are the colors just to indicate the region or does it actually change it? I'm asking because no matter what color I had it on the US games and the PAL would load.

I also notice when I was cutting the traces on the region pads JP3 doesn't connect across on the USA VA2 like the other boards I saw. I cut across the white line, but still had some continuity between the two pins.  You can look at the picture I posted above to see what I'm referring to.

SukkoPera, 2022-01-07:
The colors are [defined in the code](https://github.com/SukkoPera/MegaDrivePlusPlus/blob/master/MegaDrivePlusPlus.ino#L149). The color indicates what region the mod is trying to enforce, but if the installation was not done properly, it might not reflect the actual truth.

On your board you had to cut JP2 and 4 and remove the cap on JP1. JP3 was already open. "Continuity" means "close to 0 ohm" (let's say < 50 ohm as a rule of thumb), you will never get a fully open circuit unless you remove all the components from the board. If you get like 1 Mohm, it's effectively open.

I'm not sure what the difference between "changing the video mode" and "locking down to a region" would be. Not all games check the actual console region setting. For instance, Sonic EUR has no check at all, but we Europeans like to switch to NTSC mode so that it runs at 60 Hz as it was intended by the programmers. On the contrary, Virtua Racing is very picky about the region and will refuse to start if running on the wrong console.

sincere360, 2022-01-07:
The game I'm testing with is International Superstar Soccer Deluxe. It's the only import game I really wanted and just came in from Italy. I tested on a non modified system and its region locked as it does not load. It does load on the system I just modified, but from what you just told me I need to change one of my wires and cut the trace at JP4  I think I just got ahead of myself and was excited to play the game and didn't go through all the documentation too well the first go round. 

Also do you know why my 6 button pad does not work, but my 3 button does? I was looking through the code to see if I had to enable anything, but nothing stood out to me. 

Here is how it should be wired on my USA VA2 board based on what you just specified to me. I do have the cap removed on my actual installation. I just used this image as I'm not at home right now. I will rewire this tonight and let you know if anything changes.

[image]

SukkoPera, 2022-01-07:
That should be correct.

What does it mean that the 6-button controller does not work? MD++ does not interfere at all with the way games read controllers.

sincere360, 2022-01-07:
The 6 button controller does not respond at all, but if I use the 3 button it works. It's a new controller and it worked fine prior to the mod and still does on another system. I'll do some more troubleshooting and see what I can figure out.

SukkoPera, 2022-01-07:
You  might have shorted the controller lines while soldering the wires, that's the only reason I can think of.

sincere360, 2022-02-05:
Sorry it took so long to get back and report on this. I rewired the board according to what we had discussed and everything works as it should now. Also the controller issue was due to a short. I rewired my controller and switch my wires from 26 AWG to 30 AWG.  Thank You for the help.

closed 2022-02-05 by sincere360

SukkoPera, 2022-02-07:
Thanks a lot for reporting your findings. Well done :).

PS: AWG26 should be ok anyway.