SukkoPera/MegaDrivePlusPlus #27: Problems with IGR and mode switch using 8bitdo m30 wireless 2.4 gamepad

orabeidm, 2022-08-24:
I installed this mod today and found that it only works through the reset button on the console and does not work with the gamepad 8bitdo m30 wireless 2.4. 
2 more people from the official telegram forum of this project faced this problem (nicknames in telegram @Fanaras @TurBoss and me @DzmitryArabei)
Can this be fixed in the project's readpad.S file or something else?

Please help!

SukkoPera, 2022-08-25:
Unfortunately I don't have that controller. Unless someone can lend me one or can take some measurements with a scope or logic analyzer I cannot really do anything about it.

orabeidm, 2022-08-25:
what measurements do you need to take? I can do it myself or ask my friend who has the appropriate equipment and knowledge.

SukkoPera, 2022-08-25:
I need about a capture of at least pins 1, 6 and 9 of the controller port while a 3-button game is running and buttons UP, B, C, A and Start are being pressed and released one at a time.

The capture should be analog and include voltage levels. Ideally, it should be done with a Saleae Logic analyzer as I'm pretty familiar with it, but other solutions could work.

orabeidm, 2022-08-25:
I can buy the cheapest Chinese analyzer clone and try to record from these 3 pins. (there 8 channels supported) and send to you. The quality of the recording will most likely not be very high, but I hope this helps.

SukkoPera, 2022-08-25:
I'm not sure the cheapest analyzer would also capture analog data, so I'm not sure it would help.

orabeidm, 2022-08-25:
when a wired pad is connected, 0.5v appears on pin 9 of the connector, when C is pressed it drops to 0.25, when b is pressed, to zero, if 8 bitdos are connected, then 0.35v, and when C is pressed it drops to 0.08

on pin 6 when connecting 8bitdo is also 0.35v, when you press B it drops to zero

SukkoPera, 2022-08-25:
Those values do not make any sense. Those signals are constantly changing, they cannot be measured with a multimeter. That's why I said a scope or a LA is needed.

orabeidm, 2022-08-25:
> .

yup

orabeidm, 2022-08-27:
bought a copy of the saleae logic analyzer. it is determined by the computer under win 11 and also works in the official program of the latest version. Tomorrow I'll try to solder the analyzer outputs on the console and record the keystroke logs. I will do it for the first time.

orabeidm, 2022-08-27:
I am the dumbest person in the world!!!
the picture of the installation in the wiki shows port 2 with digit numbers. not board port1 for md2. I soldered the mod late at night and soldered it according to the picture to port 2, while according to the instructions it was necessary to solder according to the instructions to port 1! you need to take a photo of the board from below for the wiki to install the mod in md2. I will also say that this mod is not very useful for users of microsd cartridges from KRIKzz, because the main menu of the cartridge is accessed by pressing the power button off / on. when you press the reset key, only the game is already sewn into the cartridge to be restarted. if it is possible to expand the functionality of the mod with the implementation of on / off with the gamepad, it would be much more useful for all users of the flashcart.ё

SukkoPera, 2022-08-29:
You see, this is one of the problems with giving support for this project: you can't even count on the correctness of the installation.

Adding a console on/off feature is far from trivial and it would be a very invasive modification for the console.

Anyway, if you think the documentation or code can be improved or extended with new features, pull requests are welcome :).

Enjoy!

orabeidm, 2022-08-29:
oh. I'm not a programmer and I'm not good at github, I don't know how to
create a pull request. just keep in mind that the on/off functionality is
preferable to the reset functionality. no matter what  segaboard changes
requires)

orabeidm, 2022-11-26:
In addition to the fact that there is a bug with the 8 bitdo gamepad that did not go away (I will try to write logs for you), my friend found 1 more bug. in your mod, the RX0 pin is connected to the gamepad pin on which the START and C buttons are simultaneously. because of this there is a confusion in management. he also wrote his own mod with repinning and other changes like automatic polling of gamepad pins, but the problems with the gamepad from 8bitdo have not gone away. also revealed the problem of code execution on Chinese copies of arduino LGT8F328P

orabeidm, 2022-11-26:
[8BitDo M30 2.4g keypress.zip](https://github.com/SukkoPera/MegaDrivePlusPlus/files/10097210/8BitDo.M30.2.4g.keypress.zip)
frequency 25 kS/s. it was probably necessary to put 24 MS / s. I will re-measure if necessary.

SukkoPera, 2022-11-28:
I appreciate your effort very much, but unfortunately I cannot do anything useful with your work because:
1. This is not an *analog* capture.
2. The sample frequency is too slow (as you guessed, it needs to be at least 16 MS/s).
3. I have no idea what channel is connected to what.

Anyway, of course the software is well aware that the RX0 pin reports both the Start and C buttons. There are also other pins that report different buttons at different times.

orabeidm, 2022-12-03:
the original controller, when using the SEL(TH) signal, changes the data protection state in 83 ns
[image]

8 bitdo m30 gamepad has 500ns. 2.4 and bluetooth version has same delay.
[image]

good old cloned wired "16 bit" chinas gamepad  has 250ns
[image]

information from my friend who will continue to make his fork reset mod.

SukkoPera, 2022-12-04:
This is interesting. You might add some NOPs right before [this line](https://github.com/SukkoPera/MegaDrivePlusPlus/blob/master/readpad.S#L74) to make the code wait "a bit" before it samples the controller status. Every NOP will add 62.5 ns.

orabeidm, 2022-12-04:
yeah. this code work on freq 16 with 4 NOPs. on chinas arduinos analog too. with 32 freq and 17 NOPs only but unstable. my friend implemented a hard reset by connecting pin 2 of the cartridge slot. but it was not possible to implement a reset with a transition to the flashcartridge menu

pseiler, 2023-01-01:
I also face this issue. So is all the information for the 8bitdo controller provided?
I do not have a logical analyzer yet but maybe I'll buy some kinda cheap one if needed for this.

I also own those retro-bit wireless ones which I could measure then.
But for now I disconnected all the IGR wires so I do not face odd behaviour.

Are there patches I can test?

Thank you both for the effort already.

orabeidm, 2023-01-02:
This man helped me with this mod.  https://youtube.com/@user-dj1ph5bb6y .The connection pinout is different from the one used in this mod.  until he came here on github, there was no time.  I don't know if he will publish the changes here as a fork or publish his code and pinout separately.  I don't have the original files.  we communicate in a separate group in a telegram, but there are all Russians and Belarusians and we communicate in Russian.

orabeidm, 2023-01-02:
> I also face this issue. So is all the information for the 8bitdo controller provided?
> I do not have a logical analyzer yet but maybe I'll buy some kinda cheap one if needed for this.
> 
> I also own those retro-bit wireless ones which I could measure then.
> But for now I disconnected all the IGR wires so I do not face odd behaviour.
> 
> Are there patches I can test?
> 
> Thank you both for the effort already.
> 
> Also a happy new year :)

https://github.com/RangerKoc/megadrive-reset-mod

Narcotic82, 2024-07-22:
Hello, any news about this problem? I also have this controller and the button combinations do not work at all.

SukkoPera, 2024-07-22:
Not on my side, I still don't have that controller.

TimRegeant, 2025-09-28:
Glad I read this.  I was just about to install the arduino in my Genesis, but I am using a 8bitdo 2.4ghz controller.

Did anyone figure out how to get that controller to work with the igr?