SukkoPera/MegaDrivePlusPlus #28: Re-routing pad input.

DieSkaarj, 2023-01-02:
Hi there,

I'm trying to figure out how to re-route the pad read pins so that they populate D8 to D2 consecutively with the SELECT line on INT1/D3. I can't seem to find how the pins are arranged in the code. Can someone point me in the right direction?

Thanks in advance,
David

SukkoPera, 2023-01-02:
Why would you need to do that? There's a reason if pins were chosen the way they were.

DieSkaarj, 2023-01-02:
I am using RX and TX to control the LED because pins D10-13 are being used for the Overclock mod. So, moving the Pad pins along will keep things tidier.

SukkoPera, 2023-01-02:
Well, it turns out that D8 is on a different port than D0-7, which would complicate and slow down (and probably make unreliable) the polling code. There's no point in compromising functionality for "tidier" reasons, as I see it. Why can't you use D8/9?

DieSkaarj, 2023-01-02:
A', I understand now; I can. I see that was explained in the .ino file too.

It was purely for superficial reasons. Thank you for clearing that up.

closed 2023-01-02 by DieSkaarj