SukkoPera/MegaDrivePlusPlus #29: Using a clone Nano?

rfancella, 2023-01-11:
Hello All!
I have a bunch of MH-ET Tiny88 boards and would like to be able to use it in a Japanese VA4 Model 1 Mega Drive.  The board does not have a labeled TX and RX port.  Comparing the two pinouts, the RX pin on the Nano is also D0/PD0.  On the MH-ET board D0/PD0 is on Pin0 and is also labeled LED.  I wired up the board as described above using the Pin0 as "Pad Port Pin 9".  Everything works but the joypad.  The IGR and region switch does nothing.  The reset button, language, and video change as they should and the RGB LED changes color respectively.

Has anyone been able to successfully used one of these clone boards before?  If so, where does the Pad Port Pin 9 need to be connected on the board?

Here is a pinout of the MH-ET board:
[image]

SukkoPera, 2023-01-11:
RX is PD0
TX is PD1

The rest of the pins seem to use standard Arduino names, so you should be able to make the correct connections.

If the MCU used on the board is an ATmega328p (or similar), it will work.

DieSkaarj, 2023-01-11:
PD0 is also connected to the onboard activity LED. I was getting spooky action when I was playing about. It might be an idea to shift the START/C (pin 9) over to PD1 for these boards.

SukkoPera, 2023-01-11:
The led shouldn't be a problem, but is surely a choice to put it on pin 0 when practically all Arduino boards have it on pin 13.

If it is indeed a problem, I'd just unsolder the led from the board.

DieSkaarj, 2023-01-12:
I did end up trying that, but it continued to cause ghosting on hw interrupt, hah. The simpler solution was just to move the connection up one on the port.

Here's a reference to the differentials of the board compared to Arduino: https://github.com/SpenceKonde/ATTinyCore/issues/435

SukkoPera, 2023-01-12:
I have just realized that this board uses the ATtiny88, which is a new microcontroller whose existence I wasn't even aware of.

At a first glance it seems compatible enough with the ATmega328p to make this thing work, but be aware that **this is NOT a clone Nano, but rather **a different and unsupported platform, use at your own risk**.

By the way, the Tiny88 does not have an hardware UART, that is why the board has no RX/TX pins. Its pins also have different output driving capabilities, which could explain why the onboard LED was moved to pin 0 (pins 0...3 are "high sink").

DieSkaarj, 2023-01-12:
@rfancella, without any assurances you can try this layout:

```
                              +-------+                
               +--------------|  USB  |--------------+ 
               |              +-------+              | 
               | [ ] 15                        14[ ] | 
               | [ ] 16                    SCK 13[ ] | 
               | [ ] A6                   MISO 12[ ] | 
               | [ ] A7                   MOSI 11[*] | [LED Blue]
 Reset Out     | [*] A0                     SS 10[*]~| [LED Green]
 Reset In      | [*] A1                         9[ ]~| [LED Red]
 JP3/4 (50/60) | [*] A2                         8[ ] | [Single Pin LED]
 JP1/2 (JP/EN) | [*] A3                         7[*] | Pad Up (1)
 LCD SDA       | [*] A4/SDA                     6[*] | Pad Down (2)
 LCD SCL       | [*] A5/SCL                     5[*] | Pad Left (3)
               | [ ] 25                         4[*] | Pad Right (4)
               | [ ] RST                   INT1/3[*] | Pad A/B (6)
(Built in LED) | [ ] 0                         5V[*] | +5V
Pad Start/C (9)| [*] 1      VCC  MOSI GND     GND[*] | GND
Pad Select  (7)| [*] 2/INT0 [ ]  [ ]  [ ]     VIN[ ] | 
               |            [ ]  [ ]  [ ]            | 
               |            MISO SCK  RST            | 
               +--ATtiny88---------------------------+ 
```

And, in the file MegaDrivePlusPlus.ino, change line 709-712 to:

```
word buttons = (b1 & 0x38) | ((b1 & 0x02) << 2)
	             | (b2 & 0xC0) | ((b2 & 0x08) >> 2) | (b2 & 0x02)
	             | ((b3 & 0xF0) << 4);
```
The MegaDrivePlusPlus should work as intended.

With respect to SukkoPera and MegaDrivePlusPlus, I am targetting this board with my [Xiyoubu ](https://github.com/DieSkaarj/Xiyoubu)mod. It is in alpha, which is why I have ASCII art at hand ^.^

rfancella, 2023-01-12:
Thank you so much for the time you have already put into this.  :D

> I think so, but I cannot test it and don't want to support it without studying the microcontroller and board first (which I don't really have time for, ATM). Whoever wants to use this board is on their own.

rfancella, 2023-01-12:
I am on it.  I'll give it a go and report back!
I dropped in a real Nano just to prove my install is good.  It works as expected, but it was my last one.  LOL.
Please do let me know about that bootloader.  I am using this board in a couple other projects and with one, the bootup time is killing me.  LOL

> @rfancella, without any assurances you can try this layout:
> 
> ```
>                               +-------+                
>                +--------------|  USB  |--------------+ 
>                |              +-------+              | 
>                | [ ] 15                        14[ ] | 
>                | [ ] 16                    SCK 13[ ] | 
>                | [ ] A6                   MISO 12[ ] | 
>                | [ ] A7                   MOSI 11[*] | [LED Blue]
>  Reset Out     | [*] A0                     SS 10[*]~| [LED Green]
>  Reset In      | [*] A1                         9[*]~| [LED Red]
>  JP3/4 (50/60) | [*] A2                         8[*] | [Single Pin LED]
>  JP1/2 (JP/EN) | [*] A3                         7[*] | Pad Up (1)
>  LCD SDA       | [*] A4/SDA                     6[*] | Pad Down (2)
>  LCD SCL       | [*] A5/SCL                     5[*] | Pad Left (3)
>                | [ ] 25                         4[*] | Pad Right (4)
>                | [ ] RST                   INT1/3[*] | Pad A/B (6)
> (Built in LED) | [ ] 0                         5V[*] | +5V
> Pad Start/C (9)| [*] 1      VCC  MOSI GND     GND[*] | GND
> Pad Select  (7)| [*] 2/INT0 [ ]  [ ]  [ ]     VIN[ ] | 
>                |            [ ]  [ ]  [ ]            | 
>                |            MISO SCK  RST            | 
>                +--ATtiny88---------------------------+ 
> ```
> 
> And, in the file MegaDrivePlusPlus.ino, change lines 709-712 to:
> 
> ```
> word buttons = (b1 & 0x38) | ((b1 & 0x02) << 2)
> 	             | (b2 & 0xC0) | ((b2 & 0x08) >> 2) | (b2 & 0x02)
> 	             | ((b3 & 0xF0) << 4)
>                      ;
> ```
> 
> The MegaDrivePlusPlus should work as intended. However another thing to note is that there is likely a 5 second delay on boot so you may have to upload a new bootloader to the device. (I am looking into this for my own project.)
> 
> With respect to SukkoPera and MegaDrivePlusPlus, I am targetting this board with my [Xiyoubu ](https://github.com/DieSkaarj/Xiyoubu)mod. It is in alpha, but is why I have ASCII art at hand ^.^

DieSkaarj, 2023-01-12:
Here is a problem-solved that might help with the bootloader: https://forum.arduino.cc/t/burn-bootloader-attiny88-cant-be-programmed-by-usb-solved/1033045

Please, @rfancella, really do let me know how you get on with it, hah!

rfancella, 2023-01-13:
Ok, I have some feedback to start.  I hooked up everything except the pad (pins 1-7) and replaced the code, it works with the reset button.  I can change regions and a short press will reset the system. 
When I connect the wires to the pads as described in your diagram, the games starts up but the Tiny never boots.  It seem to be stuck at the bootloader.
What can I do to help diagnose this?

> @rfancella, without any assurances you can try this layout:
>
> ```
>                               +-------+
>                +--------------|  USB  |--------------+ [...]

DieSkaarj, 2023-01-13:
@rfancella, if you have a multimeter check for a short. And, if you can, take a picture of the connections and post it here. I'll see if there's anything I can recognise. The wiring in the diagram is a direct port of SukkoPeras Arduino layout, except for the start button, which has been shifted to pin 1.

rfancella, 2023-01-13:
@DieSkaarj 
Ok, no shorts across any pins on the controller port:
[image]

Top of the board:
[image]

Bottom:
[image]

Side:
[image]

The only changes in code are here:
[image]

and:
[image]

DieSkaarj, 2023-01-13:
Lookin' good to me, too, @rfancella. The next step is to disconnect everything and start introducing things back individually. However, if you're confident the issue is with the controller try just removing those lines first and then reintroduce them, one by one, back to the circuit. I probably should have asked before; did you remove the controller lines and try the device again and then put them back on, and does the device still power up through USB?

rfancella, 2023-01-13:
@DieSkaarj 
I started with not having the controller lines attached.  At that point the reset button worked to change regions/language.  Also the RGB LED also changed and saved modes appropriately.  Once I soldered in the controller lines, it just flashes the light on the Tiny like it's in bootloader mode.  If I disconnect the controller lines, it works again.
So, I will start removing the controller lines one at a time and test.  ;)

I decided to use the "cut traces" method on mine, the one I am testing with.  So the reset button is out of the circuit.
[image]

rfancella, 2023-01-13:
@DieSkaarj 
It is for sure pin 2.  I remove pin 2 and it boots just fine.

DieSkaarj, 2023-01-13:
Pin 2 was connected to pad pin 7 on the motherboard, @rfancella?

And, did you try turning the machine on with, and without, the controller attached? (not that it should change the behaviour but just wanting to make sure there wasn't anything funny going on there.)

tbh, I'm stumped. Pin 2 should connect to the SELECT signal from the controller and it looks fine from the pictures you've posted. You can test the line, but it's probably fine.

The image for the other bit of code you changed doesn't show up, can you post it again?

Now, we know it won't get the controller to work but try writing the original firmware (with your LED changes) to see if it boots and then again with the hotfix I suggested.

rfancella, 2023-01-14:
@DieSkaarj 
Yes, Pin 2 is connected to pad pin 7.
The controller attached or not makes no change in the outcome.  :(
I re-uploaded the picture of the code into the post above.
I'll try the original code and report back.  :)

> Pin 2 was connected to pad pin 7 on the motherboard, @rfancella?
>
> And, did you try turning the machine on with, and without, the controller attached? (not that it should change the behaviour but [...]

SukkoPera, 2023-01-16:
Just my own 2 cents: you can get a "real" Arduino Nano clone for ~4€, which would make things much easier :).

rfancella, 2023-01-16:
@SukkoPera 
Agreed.  This is now an exercise in education for me.  ;)  And something to do with these POS clone boards.  LOL

> Just my own 2 cents: you can get a "real" Arduino Nano clone for ~4€ shipped from China, which would make things much easier :).

DieSkaarj, 2023-01-16:
They're not so bad tbh. Happy to hear about the performance with a change in bootloader. Thanks for letting me know, @rfancella.

I've got the same controller layout on my mod Xiyoubu to the one I suggested and it works flawlessly.

I'm really baffled as to what's going on with your board , @rfancella. Did you try rewriting the original MD++ firmware with the select wire on?

SukkoPera, 2023-01-16:
Note that on this board the USB connection uses PD1 (D+) and PD2 (D-). I don't know the USB protocol well enough but maybe SELECT being connected to PD2 might interfere with the bootloader startup?

You'd better move SELECT to another pin, but you need one with EXTINT capabilities or to switch to PCI NTs and make substantial amendments to the code.

DieSkaarj, 2023-01-16:
https://github.com/DieSkaarj/MegaDrivePlusPlus - branch mh-attiny88

I think I got all references to INT0. If it works I'll better annotate the changes and send @SukkoPera a pull request.

rfancella, 2023-01-16:
@DieSkaarj 
Let me know when your branch is ready and what changes I need to make and I'll do a pull and test.

>

DieSkaarj, 2023-01-16:
It's good to go, @rfancella.

DieSkaarj, 2023-02-12:
Hi @rfancella,

Did you have any success with this board? I'm having trouble reading START consistently and I think it might be related to this issue.

Thanks,
David

rfancella, 2023-02-13:
> Hi @rfancella,
> 
> Did you have any success with this board? I'm having trouble reading START consistently and I think it might be related to this issue.
> 
> _**UPDATE**_
> 
> The issue stems from the device not being able to read -5V (a bipolar signal) which can lead to all sorts of weirdness. The solutions that I've yet to test are either use an OpAmp or use a DC offset.
> 
> Thanks, David

David,
I have not.  I got a handful of consoles in for repair and have not had a chance to look at it again.  I am hoping to get mine back out later this week and try it with a new board.  :dagger: 
Thanks,
Ron

SukkoPera, 2024-03-02:
Closing due to inactivity after 1Y, feel free to reopen if necessary,

closed 2024-03-02 by SukkoPera