SukkoPera/MegaDrivePlusPlus #30: Sketch cannot compile on Nano Every

jon-nielsen, 2023-05-01:
Hello

Thanks for the project!

Unfortunately I have a compatibility bug to report.

The sketch cannot compile using the IDE with an Arduino Nano Every board.

The error is in the readpad.S file. "Error: Constant value required" on several lines of code.

If this cannot be fixed I suggest updating readme with a note.

Cheers

SukkoPera, 2023-05-01:
I don't know anything about the Nano Every, sorry. I hope someone else will contribute a fix.

jon-nielsen, 2023-05-01:
I may try with an Uno instead. Thank you.

SukkoPera, 2023-05-03:
I'd recommend a $3 Nano clone from Aliexpress.

SukkoPera, 2023-05-11:
Only replacing switches. I think someone made a board which also includes a clock generator controlled by this project but can't say more than that.

L10N37, 2023-05-11:
you can get the colour burst straight off the MCU without another Clock gen board ...I know you could with 32mhz. It's been a while since I checked docs. I may fork the repo in future to suit LGT8F328P (32Mhz) and set a fixed clock out so the board has the added functionality of the QXO-1100 or similar , all in one.

only I can't get the megadrive working I planned on using for the project. Paid way too much for it but I usually get the faulties working pretty easily. This one likely has a dead component and needs more analysis and a proper probing. If I get around to It, it will happen :P

closed 2026-11-07 by jon-nielsen