SukkoPera/MegaDrivePlusPlus #31: Buttoncombos not working: NANO USB C clone

Lo0twig, 2023-12-14:
I can neither switch the region nor trigger the IGR.
I have checked the wiring several times and it seems to be correct.

The L LED stays on after you press A and goes off again after a "cooldown" the next time you press A.
The same applies to the B button.
The controller inputs do not seem to be read correctly.

If I remove all cables for the controller inputs, switching via the reset button and the reset function no longer work.

SukkoPera, 2023-12-15:
Are you using an Arduino Nano? Make sure the fuses are set correctly.

Lo0twig, 2023-12-15:
Yes I am using an Arduino Nano clone with usb c. I had to move the Rx led and resistor so 6button controllers work. But the behavior is the same no matter if these are removed or not.
[image]

Lo0twig, 2023-12-15:
Uploading IMG_9630.MOV…

SukkoPera, 2023-12-15:
Is it a clone with an ATmega328?

Lo0twig, 2023-12-15:
Yes

SukkoPera, 2023-12-15:
Well, I'm sorry but I don't know what to suggest. The Reset button stuff is independent of the controller snooping, so it should work even without the controller wires. Triple check the wiring and maybe post a pic of your installation so that we can see what you did.

Lo0twig, 2023-12-15:
[image]
[image]

Lo0twig, 2023-12-15:
[image]

Lo0twig, 2023-12-15:
[image]

Lo0twig, 2023-12-15:
https://github.com/SukkoPera/MegaDrivePlusPlus/assets/44567649/2ad0ee81-8f54-4ce6-9240-450361c0eb3a

SukkoPera, 2023-12-15:
Wiring looks correct, just make sure +5V and GND are actually what they should be.

Also try another game, many games do not poll the controller at all times (in particular not during the initial screens), and since the Arduino is only snooping passively, it must wait for the game to do it.

Lo0twig, 2023-12-15:
Pulled 5v before from an other point. No difference.
Also tried several Games via Mega EverDrive Pro. Same behavior.

SukkoPera, 2023-12-15:
I have no more clues, sorry :(.

Lo0twig, 2023-12-15:
😢