SukkoPera/MegaDrivePlusPlus #32: video links are all broken

L10N37, 2024-01-20:
I'd like to get a good overview of the installation, especially the RGB LED to indicate modes. I can't see much about it other than the Pinout showing the normal power LED is for some reason powered by the MCU? 

Are the videos available at all? 
It's a boxed system so I don't want to be doing and trial and error, just a smooth , trouble free installation along with 3BP

SukkoPera, 2024-01-20:
The videos were kindly provided by Dermot Sweeney. He has now renamed his channel to Game Tech Refuse. I see he no longer has those videos but has [this one](https://www.youtube.com/watch?v=8vtyRr0Hqp0). Hope it helps.

I will remove the dead links anyway, thanks.

L10N37, 2024-01-28:
Thanks, I am very confused about your model 2 diagram after playing around with a model 2 I have given up on (triple bypass has terrible signal to noise ratio now and video is way over driven) 

I did play around with my own into sketch and make something to capture button inputs without looking at anyone else's code, and I stuffed about with multimeter and code trying to force a reset.

Your diagram/s show lifting of the legs toward front of console and soldering reset out to them. This makes no sense. Reset on model 2 is by sending a logic high with enough current across to the rear legs. 

the reset in pictured just buzzes straight back to the pads the lifted legs of the reset button were oiginally soldered to. I can understand the reset in wanting to sense the active state of the reset line, but soldering the reset out back to the lifted leg(s) leaves a floating connection that goes no where.

in my testing jumping 5v from anywhere on the console to the rear legs will cause a reset, even thru a resistor.. but 5v off a GPIO pin on nano just gets sunk by the reset line and does nothing.

SukkoPera, 2024-01-28:
First of all, about the videos: I wrote an email to Dermot last week but I got no reply so far, thus I am going to remove them for real.

Regarding your problem, I think you are a bit confused: IIRC and unless you have a weird board, reset is active-low on the MD2. The back legs are just ground, and that is why the console resets when you put put 5V on them: you are creating a short circuit on the power supply!

[image]

The wire you are supposed to connect to one of the LIFTED FRONT legs is Reset IN, not OUT! It is the ORANGE wire in the pic above. The Arduino will pull it up, sample it and reset the console when it goes LOW, i.e. when the button is pressed and the front legs get short circuited to the back legs.

Reset OUT goes is the BLUE wire, you can see from the pic where it is supposed to go.

Hope it helps.

SukkoPera, 2024-01-28:
Video links removed.

closed 2024-01-28 by SukkoPera

L10N37, 2024-01-28:
I am very confused, the lifted legs are floating. They can not do anything at all, other than float - unless the button is pressed physically which would connect them to the other 2 legs plane. 
How can it sample anything from a floating connection? There is no connection to the mainboard on them when lifted.

The blue wire is connected to the same plane the lifted legs used to be connected to, just confirmed with multimeter. This is also not making sense.

SukkoPera, 2024-01-28:
Exactly: the Arduino has in internal pull-up resistor which will prevent the floating and keep it high. When you press the button, it goes low and the Arduino knows you pressed the button.

The blue wire will go to the place that used to be grounded by the button press before you lifted the legs, and it will now drive it as required (i.e.: keep it high and bring it low when it must trigger a reset).

L10N37, 2024-01-28:
Yes I thought the chip could reset the console with a button combo is all, not sample physical presses , i guess all of this was to make it work with both active high and low consoles.

SukkoPera, 2024-01-28:
No, it's to enable region switching through the reset button: if you hold it for a few seconds you will switch the region and NOT reset the console. This implies that the button but be disconnected from the reset line and that must be handled by the Arduino.

L10N37, 2024-01-28:
Ok, I get it now. Thanks for clarification. 
Arduino doesn't seem to be able to reset megadrives, at least not model 2. 
If this megadrive is ever sorted, I will use this - region free mode on the cheap ED clone still gets blocked by a lot of US titles. 
It's just the scalers being out of wack with refresh rate, and the maintaining composite and tidying it up is why I was doing something on the side. This will suffice, NTSC snes has incorrect refresh rate and that was fine on my OSSC. Trying out GBS-C now (mainly a crt person for old stuff, but starting to enjoy some stuff on LCD)

Thanks :)

SukkoPera, 2024-01-28:
Unless you have a broken Arduino, the only possible reasons can be bad wiring or code alterations. There are quite a few people around happily using this. Good luck!

L10N37, 2024-02-10:
please note when I said I couldn't get the console to reset I meant with other code. I have tried 3 microcontrollers and it's driving me nuts. I've also fried a few GPIO pins. I finally went through your code and I can't see anything different for your reset other than  using the fast write library.

You sample the rear legs to sense active state, disconnect them so you can control when to send that signal across to the front legs to perform a reset (or change modes) which happens to be when you don't just press it quickly (hold to switch modes, short press reset)

anyways, let's put all that aside as it's not relevant .. but sending a high to the front legs will not reset my JP model 2. Not even with a single gate buffer. Senses active high from rear legs, send high to front legs. nothing. It just goes from 5v to 0v on my multimeter and sinks the current, again, even with a buffer. 

Even if I set the pin perma high and tap it. It just sinks it.

If I leave it high and soldered on it just kills my GPIO pins. 

I can sense controller combos using port states D2 to D7 and just mask off the D0 and D1 RX TX pins, then I use a port B pin for the select line. Managed to get it working with 3 button and 6 button combos using the same polling code. Ditched the interrupts. 

I'm using SI5351 to provide NTSC / PAL clocks which is fine for dual region. I just haven't moved onto anything else because not being able to reset the console is driving me bonkers!

What the heck is going on...!

L10N37, 2024-02-10:
https://gendev.spritesmind.net/forum/viewtopic.php?t=1262

"WRES - Active high reset signal, pressing the RESET switch shorts it to +5V. Normally held low with a pull-down resistor. No debouncing."

I've had no luck finding the pulldown resistor.
Tried the actual pin labelled reset as well. 49. This just sits high at 5v 

"MRES# - Directly from the RC reset circuit (R=47K, C=10uF), goes to cartridge connector."

played with this. It's labelled 47 but it's 4k7. Not sure what's going on here TBH. Both sides of the resistor when removed are 5v. Console seems 100% functional without it. maybe this is the one that the game software has access to. Again, no luck.