SukkoPera/MegaDrivePlusPlus #33: Mega Drive 2 install, purple led, reset and controller not working

ElliottNieves, 2024-02-28:
I've followed the mega drive 2 guide as best I can, it's not super super clear but I've managed to piece it together as far as I know.

When I turn the console on, the RGB led I have wired to the red led, green led and blue led pins with 10R resistors on each, lights up purple (so either red and blue or maybe it's all 3 colours?).

The console runs at 50hz, no button combos do anything, reset button doesn't do anything either. Sonic 3 and Sonic 2 will load the title screen but when trying to actually start the game they reset.

The curious thing is that if I hit the reset button on the Arduino, the led turns off, and the console switches to 60hz for a few seconds before going back to 50hz.

Pressing the C button causes the RX light on the Arduino to light up, but no other buttons do.

I looked through all the other issues and found someone who had the same thing, where their C button would light up the RX, they said it was a faulty board because they used a clone. I was also using a clone so I have now swapped it out with an official Nano, which has had the software uploaded to it as well, and it behaves in the same way - purple led, nothing works, reset on the Arduino switches it to 60hz momentarily, game won't go past the title screen - the only difference being now not even the C button lights up the RX.

What have I messed up? I've never had this many problems with a console mod. If I can work this out I'd be happy to help update the guide to make it more clear.

Sorry about the crappy photos and crappy looking wiring - I've checked continuity on everything and can't find any issues so I assume it's something to do with the actual pin wiring or something with the software.

Some of these photos are the clone and some are the new official board. The wiring was less messy before I replaced the board as I'm just in troubleshooting mode now. Ignore the electrical tape around the resistors, that's just temporary until I get it working.

[image]

[image]
[image]

[image]

[image]

[image]

SukkoPera, 2024-02-28:
> When I turn the console on, the RGB led I have wired to the red led, green led and blue led pins with 10R resistors on each, lights up purple (so either red and blue or maybe it's all 3 colours?).

It's not related to this problem, but 10 ohm is waaaaay too low for a LED series resistor. If you don't want to do calculations, as a rule of thumb never go below 220 ohm. If you use modern leds, they are superbright and 1k or even more is the right choice. This is not something to be overlooked as it WILL destroy the Arduino pin sooner or later.

> The curious thing is that if I hit the reset button on the Arduino, the led turns off, and the console switches to 60hz for a few seconds before going back to 50hz.

This is not really curious at all: when you reset the Arduino, it loses control of the videomode/language lines for a moment, so anything can happen.

> Pressing the C button causes the RX light on the Arduino to light up, but no other buttons do.
>
> I looked through all the other issues and found someone who had the same thing, where their C button [...]

(Good) Clones are not "crappier" versions (not as long as they use an ATmega328 chip at least), so there's no reason why they should behave differently.

What you mention about the RX makes me think that you did not read the [FAQ](https://github.com/SukkoPera/MegaDrivePlusPlus/wiki/FAQ-and-Troubleshooting). Please do so and pay particular attention to points 3 and 4.

ElliottNieves, 2024-02-28:
The Mega Drive 2 guide specifically says "If you are using a dual or RGB led instead, you should put a small value resistor (10-47R maybe, the bigger, the fainter your colors will be) in series with every anode." So I'm not sure how I was supposed to know that 10R is way too low. It's easy enough to swap it out with a 1K, so I'll do that, but maybe update the guide or make it less confusing? I don't know, maybe I'm an idiot and misunderstood something but it seemed pretty straightforward as an instruction.

Okay, I meant that it's curious in that the Arduino seems to be doing _something_ when pressing it's reset, seeing as the LED is stuck on purple, the reset button is doing nothing, and the controller inputs are doing nothing.

I see the FAQ, but I did include the readpad.S file when uploading to both boards. I've reuploaded just now and double checked, and it's still functioning the same way. I didn't see the thing FAQ point about the C button until now, so thanks for pointing that out - but I still am confused as to why this isn't working.

When I removed the 5V to plug the board back into the computer while it was wired up, the LED went Red while it was powered via the USB port (which seems more like the correct behaviour), if that helps to diagnose the issue.

I'll recheck my wiring for a fourth time because now on the official board I am not getting the RX lighting up with the C button - but the LED still goes purple, no button combos do anything, the reset button doesn't work, and games reset when you try to actually play them.

I've now tried Sonic 1 as well and it just loops the SEGA screen, won't even go to the title screen. Do you have any idea what's causing this resetting?

I'm only asking here because as far as I can tell I have followed the guide as closely as possible, all my wires are in the right spot and I've checked continuity as best I can, and as far as I know the software is installed on the Nano correctly. Beyond rewiring everything once again and crossing my fingers, I'm trying to figure out if I might have missed something and there is a fatal flaw in my install. 

The fact that the guide seemed to tell me to use a 10R resistor when I shouldn't, that it includes a note that the +5V and GND points are swapped in the diagram (just fix the diagram so it's not confusing?), and that there is the addition of a second photo of someone else's install to clarify the install process for that board type, all lower my confidence that what I am reading is explicit enough to give me the best chance to wire this up correctly. I'm not saying it's wrong, but I find it hard to be 100% sure I have done this right.

SukkoPera, 2024-02-28:
Ahhh, but you mean you put that resistor in series with the built-in 120 ohm led. Then it should be fine. My bad.

The guide is what it is. This is an open project, I'll happily welcome any contributions to the guide, software, or whatever.

You can try enabling debugging mode: you will need to uncomment #define ENABLE_SERIAL_DEBUG and #define DEBUG_PAD at the top of the sketch and install the library mentioned in the FAQ. You should also put a diode in series with the +5V pin, then you can leave the PC connected, open the serial terminal and see what's going on.

A picture of the soldering you did on the bottom of the board could also help.

ElliottNieves, 2024-02-29:
I mean, the instructions for using an RGB LED are not super clear, I had to piece it together by looking at what was on the main page and the mega drive 2 page. So here's what I did and you can tell me if it's wrong:

I'm fairly certain this RGB LED is common anode not common cathode based on testing it with my multimeter, so I had to connect the anode to the right side of the original LED spot on the board, and then I put a 10R resistor on each of the 3 coloured legs and connected them to the chip according to the diagram on the main install page. Green to green LED, Red to red LED, blue to blue LED.

If there is something I've missed there, let me know. Maybe it doesn't work at all with a common anode as the guide says common cathode. I have removed the original built in LED, but again it wasn't clear from the guide to not remove it if that's what I was supposed to do.

Is incorrectly wiring the LED enough to make the entire mod not work?

I'll try debug mode and see what happens I guess.

SukkoPera, 2024-02-29:
If you miswire the LED, the worst that can happen is that it won't light up.

It only gets worse than that if you don't put series resistors on the pins, in which case you might damage the microcontroller pin driving it but even in that case, the rest of the microcontroller should still work.

So well, don't bother too much with the LED for the moment and just focus on getting the main functionality working. Debug mode will help you.

ElliottNieves, 2024-03-01:
Was trying to re-upload the program into the Nano and having connection issues, then I removed all the wires and cleaned it up, plugged it into the computer by itself and it flashed fine. I changed the code by un-commenting out the line about it being common anode because that's what my LED is.

I resoldered everything, and double checked continuity on all my wiring again, and now when I power it on the LED stays green. Better than purple I guess. But the behaviour is still the same.

Sonic 1 loops the Sega screen, sonic 2 and 3 won't go beyond the start menu, and just reset.

The only difference is now my reset button is working again, but only for actually resetting the console, not changing the modes.

Pressing reset on the Arduino still causes it to go to 60hz momentarily and then go back to 50hz.

Now, when I plug it into the computer to try and enable debugging, it won't flash again. I've installed the diode on the 5V line. It's as though being wired up to the console is causing IDE to struggle to communicate with the Nano.

It doesn't make any sense because it detects the hardware and starts the process when I press upload, and the RGB LED in the mega drive even starts flashing red when it was doing nothing before starting the upload. But I get repeated:

avrdude: stk500_recv(): programmer is not responding
avrdude: stk500getsync() attempt 1 of 10: not in sync: resp=0x00

10 times and then it fails.

Do I really have to unsolder and flash, then resolder every single time I want to change the code? What is going on here?

I'm trying to get actual debug output via serial but I can't unless it will let me flash the board.

Here are photos of the wires.

[image]

[image]
[image]
[image]
[image]

SukkoPera, 2024-03-01:
Please see FAQ point 7. Keep your console powered on for the flashing. If you still can't flash the Arduino, you have some wiring issue.

From your pictures I cannot understand if you cut the two tracks as indicated in the installation instructions or not.

Also, it doesn't appear you lifted the reset button bottom legs.

ElliottNieves, 2024-03-01:
There is nothing connected to the board. No power, no video, no controllers. It still won't flash. If you don't actually know what wires could be causing this, I would say it's an oversight. I haven't bridged any pins on the Nano and I've checked continuity to all of them. I'm going to have to just remove all the wires again and flash it, then reconnect everything.

I have cut the tracks where specified and checked continuity and lack of continuity where relevant, as well as checked continuity to the pins that the traces run to, and from the pins all the way to the Arduino. I'll remind you that I've been trying to get this working for 4+ days now and I've checked and rechecked everything.

I didn't lift the reset button legs because I did that the first time and the reset button stopped working, so I instead referred to the other Mega Drive 2 guide "PC BD MD2 VA0 PAL" because that's what my board is labelled as. On that one, it says to cut a trace and connect a wire to access reset out. Reset button now at least works, which seems more promising than before.

[image]

SukkoPera, 2024-03-01:
As I said: the Megadrive must be powered ON if you want to flash the Arduino while it is connected and the diode is in place. There's no chance for it to work otherwise. Also make sure to disconnect your controller from port 1.

After you enable debugging, you should see a message like "Reset button pushed for a short time" in the console when you press Reset, please check for that.

ElliottNieves, 2024-03-01:
My bad, I missed that detail. 

I've managed to flash the Nano with the same sketch as before but with  #define ENABLE_SERIAL_DEBUG and #define DEBUG_PAD enabled.

For no discernable reason (I haven't touched the wiring) I can now get into a game past the title screen.

When I first flashed it, holding the reset button was changing the mode with no game inserted, and it would reset the console if pressed. Once I insert a game the reset button does nothing.

There's still no controller combos working either, neither IGR or changing modes works.

I installed SendOnlySoftwareSerial by installing it as a library from the zip of the repo. I'm not certain that's the correct way to do it, but I had no errors when compiling after enabling the debugging lines of code so I assume it's okay? Let me know if there's something else I should have done as I'm unfamiliar with debugging an Arduino and the installation instructions on the repo said to just put it in a folder, which is a little too vague for a pleb like me.

Regardless, when I check the serial monitor in IDE, there are a few situations where I actually get output (resetting the Arduino, turning the console on, turning it off while the Arduino is still connected to the USB port, etc.) but all of it prints illegible garbage characters my computer can't display. I'm going to assume this is not the correct way to monitor the serial output, but again, there are no instructions for this as it seems relatively advanced and I guess the assumption is that anyone doing this would know how to get it working. What am I meant to do to get readable, usable and useful serial output?

Thanks for your help so far, this is the closest I've been to it working I think.

ElliottNieves, 2024-03-01:
More info:

I can turn the console on with no game in it and hold reset to switch modes. The LED changes colour correctly. If I wait the 5 seconds for the program to save the mode and turn it off, I can turn it back on in 60Hz mode and it kind of works.

Except for the fact that any real mega drive game I try (Sonic 1, Sonic 3 so far) says "this game is intended for use on a european mega drive" or something to that effect. If I try to boot the game in it's correct, 50Hz mode, Sonic 1 bootloops and Sonic 3 won't go past the title screen, it fades to black and resets when I press start at the title screen.

If I put in a repro of Sonic 3 Complete in 60Hz mode (Sonic 3, Sonic and Knuckles, and Sonic 3 and Knuckles combined into one ROM and flashed to a flash cart - I figured this wouldn't have region lock because it's a repro), the main title screen where you select which game works, I can load up Sonic 3, but then it won't go past the title screen again. If I pick Sonic and Knuckles however, that will get me into the game at 60Hz and works fine. But again, no button combos work, reset button does nothing, it won't even just reset the console.

What on earth is going on here?

SukkoPera, 2024-03-01:
Set serial baud rate (bottom right corner, I think) to 57600.

Please stop doing random things, they'll bring you nowhere. Let's concentrate on getting debug output.

SukkoPera, 2024-03-01:
What happens when the console is off has no relevance whatsoever. Unless you are debugging, the Arduino would be off just as well.

So please take a known working game, possibly the first Sonic, insert it, turn on the console and post what appears in the serial monitor. If it loops on "Starting up", just reset the Arduino.

ElliottNieves, 2024-03-01:
It shows the sega screen, then resets.

"Starting up..." loops until the console resets and then it prints:

01:35:07.831 -> Reset line is HIGH at startup
01:35:07.863 -> Resetting console
01:35:09.580 -> Ready!
01:35:09.613 -> Pressed: 

Then resumes looping "Starting up..."

Resetting the Arduino does nothing to change this behaviour.

SukkoPera, 2024-03-01:
As you see in the second line, the Arduino logs when it resets the console. If it's resetting without anything being output in the console, you have some other installation problem.

Please check the ground wire. Maybe it's touching the other pad of C8?

ElliottNieves, 2024-03-01:
It's resetting because the game is reaching a certain point it seems. Like a said above when I was doing "random things", different games reset when you perform certain actions. Something else that isn't an input is triggering the reset. But I will check the ground wire and report back.

SukkoPera, 2024-03-01:
Except that the Arduino has no clue of what is going on in the game...

ElliottNieves, 2024-03-01:
Ground wire is fine, definitely not touching anything it shouldn't, double checked with a multimeter.

Idk what to tell you... If I just run Sonic 3 in standard 50Hz mode, it will work fine and won't reset when starting the game (but no button combos work, reset button doesn't work), but if I then change nothing else except plugging in the USB cable to look at the debug monitor, then it resets when trying to start the game. The Arduino is the only thing changing in that scenario and the result is different...

The fact that the mod seems to work perfectly fine without a game cartridge seems fishy to me. I've looked at the cartridge slot pins and they all look fine. Not sure what else to do other than inspect the entire board very closely and see if anything is shorting?

I realise there is probably not much else you can do to help here especially if you've never encountered anyone with this specific issue before. I'm probably going to have to try and repair the cut traces on this Mega Drive and try a different mod.

SukkoPera, 2024-03-01:
One more thing we can try:

Remove all the wires you soldered to the controller port on the Megadrive, and possibly from the Arduino too.

Locate [this line](https://github.com/SukkoPera/MegaDrivePlusPlus/blob/master/MegaDrivePlusPlus.ino#L626) around line 634 in the .ino:

```
	/* Enable interrupts: we can't use attachInterrupt() here, since our ISR is
	 * going to be bare
	 */
```

and add just BEFORE that:

```
	 pinMode (0, INPUT_PULLUP);
	 pinMode (2, INPUT_PULLUP);
	 pinMode (3, INPUT_PULLUP);
	 pinMode (4, INPUT_PULLUP);
	 pinMode (5, INPUT_PULLUP);
	 pinMode (6, INPUT_PULLUP);
	 pinMode (7, INPUT_PULLUP);
```

Of course no controller combos will be detected, but at least the reset button should work.

Also there's not much point in repairing the cut traces, all IGR mods will use the same points.

ElliottNieves, 2024-03-02:
Okay, I'll give that a try.

I mean I'll repair them so the console functions normally on it's own and get another mega drive, as I suspect the issue is with the console.

ElliottNieves, 2024-03-02:
Removing the controller wire connections from the controller pins caused it to not work at all, but then after removing the connections from the Arduino it started working again. It works without the controller wires connected and games are now not resetting anymore, and I can change region with the reset button. 

Problem solved, kind of. Is there anything you can think of that I can try to fix the controller input? Or better to just cut our losses at this point?

ElliottNieves, 2024-03-02:
Also, one more question - Sonic 1 works fine in 60Hz mode, but Sonic 3 still gives me the "for use only on PAL and French SECAM mega drive systems". Is this normal behaviour or is a region check supposed to be bypassed?

ElliottNieves, 2024-03-02:
Spoke too soon... I've now tried it with a brand new everdrive and the games play but have all sorts of graphical glitches and reset randomly/crash. Real carts don't do this but I think this Mega Drive must be busted and maybe needs recapping or something. Sorry for the trouble.

SukkoPera, 2024-03-02:
> Removing the controller wire connections from the controller pins caused it to not work at all, but then after removing the connections from the Arduino it started working again.

This is very odd, so there probably lay your installation issue. Try to solder one wire at a time, check that everything is still working, and go on with the next one.

> Also, one more question - Sonic 1 works fine in 60Hz mode, but Sonic 3 still gives me the "for use only on PAL and French SECAM mega drive systems". Is this normal behaviour or is a region check supposed to be bypassed?

Start the console in PAL mode and switch to NTSC when you see the title screen. If this still doesn't work, you will have to [make sure your Arduino has the Optiboot bootloader](https://github.com/SukkoPera/MegaDrivePlusPlus/wiki/Chip-Preparation#bootloader) and possibly recompile the .ino with ENABLE_FAST_IO after installing the DigitalIO library.

> Spoke too soon... I've now tried it with a brand new everdrive and the games play but have all sorts of graphical glitches and reset randomly/crash. Real carts don't do this but I think this Mega Drive must be busted and maybe needs recapping or something. Sorry for the trouble.

Could be just the power supply, try another one.

ElliottNieves, 2024-03-03:
If anyone ever reads any of this, another discovery is that the console has issues when not installed back into its RF shielding. For some reason  putting the shielding back on fixed a lot of instability and glitching. However, I still think this console has issues as it does display minor graphical glitches at times, but consistently. Hard to know.

Since the shielding is part of the issue, in order to test the controller input one wire at a time, I would have to install a wire, put it back in the shielding, test, remove it, and repeat 7 times. I think I'm just going to stick with the working reset button. At the end of the day I'm mainly using it to play games on an everdrive so I don't actually need to play anything that's out of region, and I can deal with just picking the correct mode by holding reset.

It is a little unsatisfying to not have the mod work perfectly but this has been more difficult than my SNES SuperCIC and RGH on an Xbox 360, by far.

closed 2024-03-03 by ElliottNieves