SukkoPera/MegaDrivePlusPlus #36: IGR ONLY Code and it's necessary pinouts option pls!

DarthMotzkus, 2024-11-06:
Hi, how are you?
I loved your project, truly awesome! But for my daily use i only need the IGR function. Can you provide the code .ino file with IGR instructions ONLY, and the necessary pinouts for megadrive 3 from Tectoy (Brazil) motherboard? It's very similar to megadrive model 2 from US.
Here's a MB picture below, is a rev. 5 Tectoy MD III without SEGA CD port and internal power supply:
[image]

Thanks in advance!

SukkoPera, 2024-11-06:
Hi,

sorry if I sound harsh, but if you only need IGR, just don't use the other functions. They won't hurt in any case, and I certainly don't want to come up with another version of the code which I would then have to maintain.

Besides, there is no way I can guess the soldering points from a picture. I would need to poke here and there with a multimeter, which of course means I should have the board in my hands and unfortunately I don't own that model.

I'm leaving this open hoping that someone else can help. Sorry again.

DarthMotzkus, 2024-11-06:
@SukkoPera ok. But which is the points that control IGR? Do i need to solder the others to code work? Or just the IGR related?

SukkoPera, 2024-11-06:
If you don't need to control Video Mode and language you can avoid the wires from A2 and A3 (and of course don't cut the corresponding tracks), but all the other wires must be soldered.

DarthMotzkus, 2024-11-06:
Ok, there! You already helping me a lot, thanks!
So, let me clarify, there's a mod of IGR in console unleashed (https://www.consolesunleashed.com/guides/sega-mega-drive-model-2-in-game-reset-install-guide/) which describes the Control Pin 6 and 9 on MD model 2 reads the ABC+START already. Do I still need to wire all the 9 Pin CTRL ports with your code? I think they are reading more buttons than necessary (since 9 and 6 pins do already read your default IGR combo), due the video modes and language necessary reading. Am i wrong?

DarthMotzkus, 2024-11-06:
@SukkoPera Ok, there! You already helping me a lot, thanks!
So, let me clarify, there's a mod of IGR in console unleashed (https://www.consolesunleashed.com/guides/sega-mega-drive-model-2-in-game-reset-install-guide/) which describes the Control Pin 6 and 9 on MD model 2 reads the ABC+START already. Do I still need to wire all the 9 Pin CTRL ports with your code? I think they are reading more buttons than necessary (since 9 and 6 pins do already read your default IGR combo), due the video modes and language necessary reading. Am i wrong?
And the mod of unleashed uses only reset IN on one of bottom legs of reset button, without detached them. Why do i need to use the "reset out" too with your code, can you explain please?

SukkoPera, 2024-11-07:
Again, I might sound rude, but if you like the unleashed mod, why don't you just use that? Why get a mod with more features and dumb it down to be the same as another one with less features? I don't see the point.

With this one you need to connect all the controller port pins because it also works with 6-button controllers, and all wires are required to read those properly.

I guess that isolating the reset button from the console and connecting it to the Arduino is only required if you want to be able to switch regions through it, since obviously if it is not isolated, every press will reset the console rather than changing region. But then again, I don't want to redesign this mod, it's been a while and I don't remember all the details. There might be implications and/or side effects.