SukkoPera/MegaDrivePlusPlus #37: Can the Language be changed with button combos?

Helder1981, 2024-11-17:
Can you change English to Japanese on the fly without changing the region to Japan and vice versa? 

I also notice there is a long delay from holding the buttons before it changes regions, is this intentional? It seems like it takes roughly 5 seconds of holding the button combo before it makes the change.

The reset button changes modes pretty quickly though.

SukkoPera, 2024-11-17:
If you are in USA mode, for instance, switching to JAP mode does exactly that (and vice-versa).

But if you are in PAL mode, there is no way to switch the language to Japanese, as the PAL+JAP combo is nonstandard, as far as I know, and as such it was left out. You should easily be able to modify the code to add that, if you really need.

Mode changes should happen immediately, don't wait for the led the blink, that happens later but the mode was changed as soon as you pressed the combo.

Helder1981, 2024-11-17:
> Mode changes should happen immediately, don't wait for the led the blink, that happens later but the mode was changed as soon as you pressed the combo.

Odd as the Reset Button changes the LED right away but the button controls don't. There is no additional resistors added to any of the controller pin lines correct? I am using the vias in front of the big ASIC to tap into the buttons in my build.

SukkoPera, 2024-11-17:
Right, no resistors.

It can happen "later" if the game is not polling the controller at that time. It might happen at the startup of any game, for instance.