SukkoPera/MegaDrivePlusPlus #42: Question on common cathode rgb led for this mod

TimRegeant, 2025-09-28:
For the common cathode rgb led I am only seeing 5mm ones on Amazon.

The original led on my Genesis model 1 is 3mm iirc.

Anyone find a smaller tri-color than 5mm or is the 5mm the one to go with?

TimRegeant, 2025-09-28:
I see those are only two color and not rgb.

TimRegeant, 2025-09-28:
Hmm, I'd prefer full rgb.  I saw one youtube installation video and liked the way they had a color light for each region.

I wish you wouldn't have closed this so quick.  I am still interested to hear if anyone knows if there are rgb leds smaller than 5mm.

SukkoPera, 2025-10-01:
No, just connet red/green (or any 2 LED pins) and then change this in the code:

#define MODE_LED_EUR_COLOR {0x00, 0xFF, 0x00}  // Green
#define MODE_LED_USA_COLOR {0x00, 0x00, 0xFF}  // Blue
#define MODE_LED_JAP_COLOR {0xFF, 0x00, 0x00}  // Red

to something like:

#define MODE_LED_EUR_COLOR {0xFF, 0x00, 0x00}  // Color 1
#define MODE_LED_USA_COLOR {0x00, 0xFF, 0x00}  // Color 2
#define MODE_LED_JAP_COLOR {0xFF, 0xFF, 0x00}  // Color 1 + 2